edk2-porting/edk2-msm #429: [REQUEST] Porting windows to exynos

davidgm14059-tech, 2025-09-30:
Hi renegade project group,

i'm porting a new thing that i hope you like it, i'm porting windows to exynos 7904 (samsung galaxy a40), and if you don't believe me, check this out, it's my repository porting windows to that phone: https://github.com/davidgm14059-tech/windows-samsung-galaxy-a40, the port is unstable and it's very experimental, i'm hard working on this port, also much times i just update the repository, please note that this is in construction

can i contribute?
(i need help because i'm starting to lose hope and i need to recover myself a bit)

and i liked this project TOO MUCH that i want to contribute on it

thank you, and have a nice day.

davidgm14059-tech, 2025-10-15:
it's been 2 weeks

J0SH1X, 2025-10-17:
Well an iusse is the wrong way.

Fork the repo then clone it locally then do your edits push it to your github and then open a pull request then ill merge it in